bernhard-42/jupyter-cadquery #87: Users may need to avoid the latest jupyterlab-widgets

wzhd, 2022-08-20:
Dependency of `cad-viewer-widget`, `ipywidgets==7.7` doesn't seem to constrain the version of `jupyterlab-widgets` sufficiently.
If I'm understanding it correctly, `ipywidgets` 7.7 accepts any version of `jupyterlab-widgets` as long as it's >=1.0.0, version 1.1.1 released on Jun 23, 2022 seems to work fine.
However,  `jupyterlab-widgets` released 3.0.0 and 3.0.1 on Aug 18, which I got just by `pip install`ing `jupyter-cadquery` yesterday on a fresh Debian system. These versions may contain some breaking changes and I couldn't get the viewer to show up:

[image]

I had tried conda/mamba before, and got similar errors.
I asked a user with a working installation and compared the versions of dependencies. Manually installing the older version of `jupyterlab-widgets` seems to be the solution.
I hope this helps others avoid some debugging.

closed 2022-08-20 by wzhd

bernhard-42, 2022-08-20:
@wzhd thanks for diving into it.
As mentioned in https://github.com/bernhard-42/jupyter-cadquery/issues/85#issuecomment-1221292840 I have updated the Readme to install 1.1.1 of jupyter-widgets
I will open a ticket with Jupyterlab how to deal with that correctly

bernhard-42, 2022-08-20:
> Manually installing the older version of jupyterlab-widgets seems to be the solution.
Indeed, same conclusion on my side

wzhd, 2022-08-20:
Thank you for swiftly finding and fixing issues :smile: