bernhard-42/jupyter-cadquery #88: Latest installation takes forever

afshawnlotfi, 2022-09-02:
```
Collecting terminado>=0.8.3
  Using cached terminado-0.13.3-py3-none-any.whl (14 kB)
INFO: This is taking longer than usual. You might need to provide the dependency resolver with stricter constraints to reduce runtime. If you want to abort this run, you can press Ctrl + C to do so. To improve how pip performs, tell us what happened here: https://pip.pypa.io/surveys/backtracking
INFO: pip is looking at multiple versions of traitlets to determine which version is compatible with other requirements. This could take a while.
  Using cached terminado-0.13.2-py3-none-any.whl (14 kB)
```

It keeps trying to resolve terminado forever

afshawnlotfi, 2022-09-02:
Not exactly sure why but when I use `--use-deprecated=legacy-resolver` it works
Source: https://stackoverflow.com/a/65793484

bernhard-42, 2022-09-11:
@afshawnlotfi `terminando` is none of my dependencies. It comes with `ipywidgets` and `jupyterlab`
`cad-viewer-widget` (a dependency of Jupyter CadQuery) has both as a dependency, however, properly tagged in `setup.py`

```text
"ipywidgets~=7.7",
"jupyterlab<3.5",
```
I don't think I can do anything here

bernhard-42, 2022-09-18:
Closing this for now, since there is nothing I can do.
If it happens again, please check with `ipywidgets` and `jupyterlab`

closed 2022-09-18 by bernhard-42