bernhard-42/jupyter-cadquery #90: Is it possible to change the default view?

snapo, 2022-09-09:
Hi,
When i generate in cadquery code the view differs from the freecad view. Is it possible to adjust the perspective of the panes?
Re-assign (1) the views of the panes to my preferred axis
Re-assign (2) the iso view default coordinates

[image]

Currently i have to rotate the object to get the same behavior as in FreeCad. It works, but everytime i resize the window i have to click and drag to get the view correct and see the front. This generates a lot lot more click and i would like to prevent that.
Or are there some other ideas how i might solve this?

bernhard-42, 2022-09-11:
@snapo  From the screenshot I derive, you are using `cadquery-server`, right?
Jupyter-Cadquery shares the viewer (`three-cad-viewer`), however has nothing to do with `cadquery-server`.

Nevertheless, I think, these are three issues:
1. Resizing browser window resets camera => needs to be fixed by cadquery-server
2. A different camera position for the isometric view
3. Swap back/front with left/right orientation button view to align with cadquery workplanes

Point 1 seems to be on the roadmap https://github.com/roipoussiere/cadquery-server/issues/37
Point 2 and 3 I will look into in the next release. It is an issue in `three-cad-viewer`, however bubbles up to Jupyter Cadquery

snapo, 2022-09-11:
Hi,
Yes correct it is for the cadquery-server (i did think its the same :-( my fault).
Would you like me to close the feature request and create one in the cadquery server?

bernhard-42, 2022-09-11:
Let's keep it open to track points 2 and 3.

For the future, the dependencies are:
```
jupyter-cadquery --> cad-viewer-widget --\
                                          |--> three-cad-viewer
cadquery-server -------------------------/
```

`jupyter-cadquery`,  `cad-viewer-widget` and `three-cad-viewer` are my projects, `cadquery-server` is maintained by @roipoussiere. I'd recommend to always first open an issue with `cadquery-server` since much of the logic "around" the viewer component is implemented there.

roipoussiere, 2022-09-12:
> (Everytime i move the window, or do a change, or re-render it... i have to rotate the object with the mouse again)

Yes, after using cq-server, I confirm it's really annoying, that's why I created the issue linked by bernhard.

> Yes correct it is for the cadquery-server (i did think its the same :-( my fault).

You are not the first person to be confused here. There is [a note about this on the cq-server readme](https://github.com/roipoussiere/cadquery-server#functionning), but it might be necessary to make it more visible, maybe at the very first lines of the readme and in the project description.

bernhard-42, 2022-09-18:
@snapo v3.3.0 of Jupyter CadQuery is out and should now per default behave like FreeCAD (as you described above).
It dependes on `three-cad-viewer` [1.7.0](https://github.com/bernhard-42/three-cad-viewer/tree/v1.7.0) , which @roipoussiere would need to integrate first before this change is visible in `cq-server`

roipoussiere, 2022-09-18:
I updated three-cad-viewer to v1.7.0 in [last commit](https://github.com/roipoussiere/cadquery-server/commit/8652c9474f22c988fa8dd421a843dbdd7d678075).
You must install from source to see the changes (see readme), or wait for 0.4.1 which should be released in the next days.

bernhard-42, 2025-04-18:
this might not be relevant any more, so closing for now.
Please reopen, if still needed

closed 2025-04-18 by bernhard-42