bernhard-42/jupyter-cadquery #91: bug in stl export

roipoussiere, 2022-09-15:
I'm trying to handle stl export in cq-server.

So far I tried this:

```py
from jupyter_cadquery.ocp_utils import write_stl_file
# ...
assembly = to_assembly(*self.models, names=self.names)
write_stl_file(assembly.compound(), 'output.stl')
```

It fails with this error:

```
  File "/home/somed/.cache/pypoetry/virtualenvs/cadquery-server-Io9SCdfb-py3.10/lib/python3.10/site-packages/jupyter_cadquery/ocp_utils.py", line 248, in write_stl_file
    mesh = BRepMesh_IncrementalMesh(compound, tolerance, True, angular_tolerance)
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.BRepMesh.BRepMesh_IncrementalMesh()
    2. OCP.BRepMesh.BRepMesh_IncrementalMesh(theShape: OCP.TopoDS.TopoDS_Shape, theLinDeflection: float, isRelative: bool = False, theAngDeflection: float = 0.5, isInParallel: bool = False)
    3. OCP.BRepMesh.BRepMesh_IncrementalMesh(theShape: OCP.TopoDS.TopoDS_Shape, theParameters: IMeshTools_Parameters, theRange: OCP.Message.Message_ProgressRange = <OCP.Message.Message_ProgressRange object at 0x7fbe17cac8b0>)

Invoked with: <OCP.TopoDS.TopoDS_Compound object at 0x7fbe0662a7b0>, None, True, None
```

In the [write_stl_file](https://github.com/bernhard-42/jupyter-cadquery/blob/master/jupyter_cadquery/ocp_utils.py#L243) function, there is a call to `BRepMesh_IncrementalMesh`:

```py
mesh = BRepMesh_IncrementalMesh(compound, tolerance, True, angular_tolerance)
```

Where `tolerance` is defaulted to `False`, but it's supposed to be a required float.

Maybe you have to set the default tolerance to a float in the function signature?

roipoussiere, 2022-09-15:
Anyway I used CadQuery builtin export function, so you have time to investigate. :)

bernhard-42, 2022-09-15:
I'd recommend to use the CadQuery functions.

Nevertheless, `write_stl_file` works as

```python
from jupyter_cadquery.ocp_utils import write_stl_file
from jupyter_cadquery import Part

box = cq.Workplane('XY').box(1, 2, 3).edges().fillet(0.1)

write_stl_file(
    box.objects[0].wrapped,
    "test.stl",
    tolerance=0.1,
    angular_tolerance=0.3,
)
```

bernhard-42, 2022-09-15:
Since each object depending on size needs different params (a tradeoff between smoothness and time needed for tessellation, `write_stl_file` doesn't have default values for `tolerance` and `angular_tolerance`.

roipoussiere, 2022-09-15:
I understand, but why in [this line](https://github.com/bernhard-42/jupyter-cadquery/blob/master/jupyter_cadquery/ocp_utils.py#L243) you set `tolerance` to `None`instead something like `0.1`?

bernhard-42, 2022-09-15:
Fair point. I should change the defaults.
I am anyhow currently preparing a new release (with isometric view and orientation buttons working like in FreeCAD or Onshape). Will change it

bernhard-42, 2022-09-17:
@roipoussiere just published 1.7.0 of `three-cad-viewer`
The reason why I increased the minor version is that it changes the default orientation behavior:
- up="Z" now uses the isometric orientation and button behaviour of FreecAD and Onshape, ...
- up="Y" works how Fusion 360 works when up is set to Y
- up="L" (legacy) works as Jupyter CadQuery before.

Will now start to publish the dependent package and jupyter cadquery
I think that addresses the comment in discord about wrong orientation.

bernhard-42, 2022-09-18:
@roipoussiere V3.3.0 is out. It fixes this issue by using 0.01 and 0.2 as defaults in `exportSTL` (module `exports`)
`exportSTL` wraps `write_stl_file` to create a compound before. This is the routine that should be used, `write_stl_file` is an internal function and shouldn't be used.

bernhard-42, 2025-04-18:
Might not be relevant any more, but for reference, please use export functions of cadquery and build123d

closed 2025-04-18 by bernhard-42