bernhard-42/jupyter-cadquery #92: Borders on viewer when using Docker standalone viewer

thejevans, 2023-01-02:
I use the standalone viewer side-by-side with vim on my laptop, and screen real estate is at a premium. There are borders around the viewer that take up unnecessary space on the screen. This occurs in both firefox and chromium.

[image]

bernhard-42, 2023-01-02:
There is a glass mode available which makes the tree transparent and shrinkable.
Can you try to add -g to the start command (assuming you use the docker command provided by jupyter cadquery)
Does it work and would it help?

thejevans, 2023-01-02:
That certainly helps, so thanks for that! Do you know what is causing the borders, though? I would be more than happy to write in an extra option to remove them if you could point me in the right direction.

bernhard-42, 2023-01-02:
I don't fully understand your setup since there is no identifiable application window in your screenshot.
Where is the viewer actually running in? What is the right side window?

thejevans, 2023-01-02:
The right side window is firefox with a custom stylesheet to hide the interface. Here is what it looks like in chromium. As you can see, the borders are still there
[image]

thejevans, 2023-01-02:
`docker run -it --rm -p 8888:8888 --name jcq -p 5555:5555 bwalter42/jupyter_cadquery:3.4.0 -v -d -g -w 1200 -h 1300`

This is the command I'm using, but I observe the same behavior with no flags

bernhard-42, 2023-01-02:
Just checked. It is the jupyterlab defaults

```
jp-Cell 
    padding: var(--jp-cell-padding)

jp-Notebook
    padding: var(--jp-cell-padding)

body
    margin: 8px
```
with `--jp-cell-padding = 10px`

bernhard-42, 2023-01-02:
When I override these three css settings with `0 px` the borders vanish
Do you have an idea how to inject these?

bernhard-42, 2023-01-02:
And there is 
```
.jp-Cell-outputWrapper
    margin-top: 5px
```

bernhard-42, 2023-01-02:
And finally my own top margin 

```
.jp-Cell-outputWrapper
    margin-top: 5px
```

bernhard-42, 2023-01-02:
The tricky part is that this is not done by [three-cad-viewer](https://github.com/bernhard-42/three-cad-viewer) (the actual Javascript component), but by the [voila](https://github.com/voila-dashboards/voila) viewer

bernhard-42, 2023-01-02:
You could try to patch `/home/cq/viewer.ipynb` in the container to override the CSS settings from this notebook.
Not yure this would work, but it would be my start point.

thejevans, 2023-01-02:
adding an html magic cell at the start of `viewer.ipynb` with these contents seems to be the minimum requirement to remove the borders

```html
%%HTML
<style>
body {
  margin: 0px !important;
  padding: 0px !important;
}
.jp-Cell {
    padding: 0px
}
.jp-Cell-outputWrapper {
    margin-top: 0px !important;
}
.tcv_cad_viewer {
    margin: 0px
}
</style>
```

closed 2023-01-04 by thejevans

bernhard-42, 2023-01-04:
@thejevans I have published a new version of Jupyter CadQuery (was anyhow planned) 
This is how it looks now in a browser
[image]

thejevans, 2023-01-04:
Awesome!