bernhard-42/jupyter-cadquery #93: Syntax error when setting darkmode

yvanblanchard, 2023-01-25:
Hello,

I followed the recommendations for a windows install with conda, but I got a syntax error when setting the darkmode them:

[image]

Could you please help ?
Thank you.

bernhard-42, 2023-01-25:
Please remove the "[1]:" at the beginning, you've copied too much

bernhard-42, 2023-01-25:
I will change the Readme and remove the ipython prefix "[1]:"  in the next release

yvanblanchard, 2023-01-25:
Ok thank you!
sorry for my silly question :-)

I also noticed that we must pay attention at indentation as well
[image]

closed 2023-01-25 by yvanblanchard