bernhard-42/jupyter-cadquery #95: Invalid cadquery version

iovxw, 2023-03-15:
I'm trying to install jupyter-cadquery with poetry:

```
poetry add jupyter-cadquery --extras=prod
```
warn:
```
Invalid requirement (cadquery (==master) ; extra == 'prod') found in jupyter-cadquery-3.5.2 dependencies, skipping
```

bernhard-42, 2023-04-27:
Sorry for the late reply.
cadquery was always a difficult dependency since it does releases very rarely, and folks are used to installing the master from github. So binding to 2.2.0 means to have soon a pretty outdated cadquery I fear.

How about removing the `extras=='prod'` and install cadquery separately, either 2.2.0 or master, as is convenient for you?

iovxw, 2023-04-28:
I don't actually use cadquery, I am using build123d and noticed that there was a missing dependency (specifically, I'm using vscode + jupyter-cadquery + build123d, it works perfectly. I prefer to display the model in the cell, so I didn't use vscode-ocp-cad-viewer), so I'd like to know the correct way to install it, it would be great if there were a documentation

bernhard-42, 2023-04-28:
Unfortunately, even when you use build123d, jupyter cadquery needs cadquery.
When the VS Code extension is finished , I will port jupyter cadquery to the underlying package (ocp_tessellate) which will make things easier. Up to now the only way jupyter cadquery got it installed with poetry is

```bash
poetry init
poetry add ipython jupyter-cadquery
poetry add cadquery-ocp=7.7.0 cadquery=2.2.0
poetry add git+https://github.com/gumyr/build123d.git
```

The "invalid constraint" message can be ignored as far as I see.

closed 2023-04-29 by iovxw

jsirois, 2024-05-08:
> The "invalid constraint" message can be ignored as far as I see.

@bernhard-42 it can't always. Would you be open to actually fixing the constraint to be a valid Python version specifier (https://packaging.python.org/en/latest/specifications/version-specifiers/#version-specifiers)? If by `cadquery==master` you meant `cadquery @ git+https://github.com/CadQuery/cadquery@master`, that would be better to say, since its valid and `==master` is not.

If you are open to this I'll send up a PR.

jsirois, 2024-07-08:
@bernhard-42 one last ping on the [above](https://github.com/bernhard-42/jupyter-cadquery/issues/95#issuecomment-2101053262).

With modern Pip, the package is now uninstallable. You're forced to use older Pip if your project is in the resolve mix:
```
python -mvenv example.venv
example.venv/bin/pip -q install -U pip
example.venv/bin/pip --version
pip 24.1.2 from /tmp/example.venv/lib/python3.10/site-packages/pip (python 3.10)

example.venv/bin/pip install jupyter-cadquery==3.5.2
Collecting jupyter-cadquery==3.5.2
  Downloading jupyter_cadquery-3.5.2-py3-none-any.whl.metadata (1.4 kB)
WARNING: Ignoring version 3.5.2 of jupyter-cadquery since it has invalid metadata:
Requested jupyter-cadquery==3.5.2 from https://files.pythonhosted.org/packages/79/ed/bed94975287784fb132f76764514e3b8d7c286ceec7610f1b1599e335167/jupyter_cadquery-3.5.2-py3-none-any.whl has invalid metadata: Expected matching RIGHT_PARENTHESIS for LEFT_PARENTHESIS, after version specifier
    cadquery (==master) ; extra == 'prod'
             ~^
Please use pip<24.1 if you need to use this version.
ERROR: Could not find a version that satisfies the requirement jupyter-cadquery==3.5.2 (from versions: 0.9.2rc0, 0.9.2rc1, 0.9.2rc2, 0.9.2rc3, 0.9.2rc4, 0.9.2rc5, 0.9.2, 0.9.3, 0.9.4, 0.9.5, 1.0.0, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.1.0rc0, 2.1.0rc1, 2.1.0, 2.2.0rc0, 2.2.0rc1, 2.2.0rc2, 2.2.0rc3, 2.2.0, 2.2.1, 3.0.0rc0, 3.0.0rc1, 3.0.0rc2, 3.0.0rc3, 3.0.0rc4, 3.0.0rc5, 3.0.0, 3.1.0rc0, 3.1.0rc1, 3.1.0rc2, 3.1.0rc3, 3.1.0rc4, 3.1.0, 3.2.0, 3.2.1, 3.2.2, 3.3.0, 3.4.0, 3.5.1, 3.5.2)
ERROR: No matching distribution found for jupyter-cadquery==3.5.2
```

Note the message from Pip "...Please use pip<24.1 if you need to use this version.".

bernhard-42, 2024-07-08:
Sorry, I missed the other message. I focussed my work pretty much on OCP CAD Viewer in the last months.
But after the next release of OCP CAD Viewer (which includes heavy changes to ocp-tessellate and three-cad-viewer) I plan to update this project (plus Jupyter 4)
Happy top accept a PR, since you seem to know what should be changed

jsirois, 2024-07-08:
Excellent. The one thing I need to know to do the change is what you intended by `==master` here:
https://github.com/bernhard-42/jupyter-cadquery/blob/578b430af6e03ba05187e4fa22f64624b58a9cf5/setup.py#L33

Did you mean the master branch tip of the cadquery project? That's spelled not `cadquery==master` but `cadquery @ git+https://github.com/CadQuery/cadquery@master` and I can make that change. If you meant something else by `==master`, let me know.

Examples:
```
# Existing syntax:
pip wheel --no-deps "cadquery==master"
ERROR: Could not find a version that satisfies the requirement cadquery==master (from versions: 0.1.0, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.5, 0.1.7, 0.1.8, 0.2.0, 0.3.0, 0.4.0, 0.4.185, 0.4.186, 0.5.1, 0.5.2, 1.0.0, 1.1.0, 1.2.0, 1.2.1, 1.2.2, 1.2.4, 1.2.5, 2.2.0b0, 2.2.0b1, 2.2.0b2, 2.2.0, 2.3.0, 2.3.1, 2.4.0)
ERROR: No matching distribution found for cadquery==master

# Working syntax to grab a wheel of the master branch:
pip wheel --no-deps "cadquery @ git+https://github.com/CadQuery/cadquery@master"
Collecting cadquery@ git+https://github.com/CadQuery/cadquery@master
  Cloning https://github.com/CadQuery/cadquery (to revision master) to /tmp/pip-wheel-oj5qqrzd/cadquery_36b3cbe08f4a47379ff7ca77c59d6dde
  Running command git clone --filter=blob:none --quiet https://github.com/CadQuery/cadquery /tmp/pip-wheel-oj5qqrzd/cadquery_36b3cbe08f4a47379ff7ca77c59d6dde
  Resolved https://github.com/CadQuery/cadquery to commit edabe5e642dc6f74a98652eefcb86c32d76ecb72
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: cadquery
  Building wheel for cadquery (setup.py) ... done
  Created wheel for cadquery: filename=cadquery-2.5.0.dev0-py3-none-any.whl size=157323 sha256=9dd9092a5f24b240b7a4f111ff7f4ac588db8721f497af0ea94c2ed1993ecbe9
  Stored in directory: /tmp/pip-ephem-wheel-cache-vd4hxpul/wheels/6d/1f/a1/56a33fbf4e1e6c42dd6be1a18a5f286e2c21a249e7a93e22f6
Successfully built cadquery
```

bernhard-42, 2024-07-08:
Since cadquery for a long time did not do proper releases, for conda we all used `cadquery=master`.
I never used pip to install cadquery, since OCP wasn't available for a long time. I actually fear that I never tested the pip `prod` install clause, to be honest. Might be broken forever :( 

What it should achieve was to install the latest cadquery master from github, as you expected, hence your `@master` seems to be the correct way

bernhard-42, 2024-07-08:
Trust that you tested it and quickly merged it

jsirois, 2024-07-08:
Thanks @bernhard-42. I did test by hand and that exercise is now included in the PR comments. I did just re-read the spec I linked in that PR though and noticed:

> Public index servers SHOULD NOT allow the use of direct references in uploaded distributions. Direct references are intended as a tool for software integrators rather than publishers.

So it may be the case that you won't be able to upload the next release to PyPI. If so, you'll need to come up with another idea for the prod extra. You should be able to use the testing exercise notes I left in the PR though to verify your prod extra changes actually work (i.e: `pip wheel -w /tmp/whls .[prod]`).

bernhard-42, 2024-07-08:
Thanks for the info, I will check when I create the next version.
Since cadquery now does have a version on pypi (https://pypi.org/project/cadquery/) I might just use `"prod": {"cadquery"}`

PR #106 (2024-07-08): Fix prod extra cadquery requirement syntax.
Previously this used `==master` as the version specifier which either
matches no versions in old, lax versions of Pip, packaging and other
Python ecosystem tools or else fails parsing altogether in new, strict
versions of those tools. We now use a direct URL VCS requirement to
indicate the tip of the cadquery master branch.

See: https://packaging.python.org/en/latest/specifications/version-specifiers/#direct-references

Fixes #95
commit aae2999b (2024-07-08): Fix prod extra cadquery requirement syntax.

Previously this used `==master` as the version specifier which either
matches no versions in old, lax versions of Pip, packaging and other
Python ecosystem tools or else fails parsing altogether in new, strict
versions of those tools. We now use a direct URL VCS requirement to
indicate the tip of the cadquery master branch.

See: https://packaging.python.org/en/latest/specifications/version-specifiers/#direct-references

Fixes #95