bernhard-42/jupyter-cadquery #98: how to change default left-right view in replay mode.

jbernalr, 2023-09-01:
Im enjoying Jupyter-cadquery a lot, in particular I like the replay mode, inlined in the Jupyter notebook. 
when im not in a large display, I would like to change the default layout of the window showing the multi select box and the figure itself. Not left-right, but first showing the replay list then the figure. 
I tried changing the "anchor" parameter of show. This seems to have only an effect using the separate cadquery 
im sure im missing obvious. could you point me please to it?

jbernalr, 2023-09-01:
```
import cadquery as cq
from jupyter_cadquery import show, set_defaults, open_viewer
from jupyter_cadquery.replay import enable_replay, disable_replay, reset_replay, get_context, replay, Replay, _CTX
from jupyter_cadquery.cad_objects import to_assembly

set_defaults(
    cad_width=640, 
    height=480, 
)
```

```
l = 100.0  # Length of the block (red x axis)
m = 100.0  # Height of the block (green y axis)
n = 100.0  # Thickness of the block (blue z axis)

# base cube
result = cq.Workplane("XY").box(l, m, n)

use_replay = True

if use_replay:
    enable_replay(False, False)
    reset_replay()
    show_object = replay
else:
    disable_replay()
    show_object = show(
        box,
        viewer="CadQuery",
        anchor="split-bottom",
        cad_width=340,
        tree_width=150,
        height=480,
        theme="light",
        pinning=False
    )

show_object(result)
```
[image]

bernhard-42, 2023-09-03:
This is not supported.
Is there a reason why you do not use the sidebar?

closed 2026-09-10 by bernhard-42