bernhard-42/jupyter-cadquery #99: Render with Blender

westurner, 2023-09-08:
Render with Blender in addition to three.js.

Blender does not have the same polygon scaling limits with complex assemblies as three.js, which works in a notebook without a kernel.

Maybe a Blender tab that doesn't render by default unless a config option is set?

- https://pypi.org/project/blender/
- ipyblender: https://github.com/kolibril13/ipyblender-experimental
  - ipyblender wraps the new Python Blender API
  - https://github.com/kolibril13/ipyblender-experimental/blob/main/src/ipyblender_experimental/__init__.py
  - screencast of ipyblender:
    https://twitter.com/kolibril13/status/1699790198505353259

westurner, 2023-09-08:
- https://github.com/ManimCommunity/manim/issues/3362 :

TIL about o3de, too. I had heard of Blender, but then I found o3de while looking for information about PhysX Blender; o3de does support physx.

> o3de:
> * src: https://github.com/o3de/o3de/
> * docs: https://www.docs.o3de.org/docs/
> * web: https://o3de.org/
> * already supports PhysX
>   * https://www.docs.o3de.org/docs/learning-guide/tutorials/physx/
> * https://www.docs.o3de.org/docs/api/gems/
>   * https://www.docs.o3de.org/docs/api/gems/ros2/ !

bernhard-42, 2025-04-18:
Thanks for the info, but I don't have the bandwidth for more rendering engines

closed 2025-04-18 by bernhard-42

westurner, 2025-04-18:
FWIW these days wgpu (wgpu-py,  xgpu) has portable WebGPU, which works in
WASM, too

https://github.com/gfx-rs/wgpu-native#bindings

wgpu-py:

https://github.com/pygfx/wgpu-py/

wgpu-py docs > Support for Jupyter lab and notebook:
https://wgpu-py.readthedocs.io/en/stable/gui.html#support-for-jupyter-lab-and-notebook

https://anaconda.org/conda-forge/wgpu-py

https://github.com/pygfx/pygfx

pygfx (wgpu) > Gallery:  https://docs.pygfx.org/stable/_gallery/index.html

And then physics, though:

- mujoco > Renderer API:
https://github.com/google-deepmind/mujoco/issues/2487#issue-2905702466 ;
filament, LuisaRender & Genesis-AI ;

- part libraries: cadquery, build123d, bd_warehouse, PartCAD,
https://build123d.readthedocs.io/en/latest/external.html

Game engines support import from e.g. Blender and somehow STL

What would physical simulation be useful or necessary for with procedural
CAD like cadquery and build123d?

- automated design verification unit and integration tests
- material strength modeling
- stress-test simulation
- Finite Element Method analyses
- rollover risk
  - (rule of thumb: when the center of mass crosses the hull of the ground
contact points, and there aren't wheels,)