bernhard-42/jupyter-cadquery #105: issues with installation

jgunstone, 2024-04-09:
hi there, 
I've been playing cadquery and build123d and think they are awesome and am excited to get it into jupyter. I've installed successfully in the past but just tried again and am having issues. 

this failed:
`mamba create -n jcq3 -c conda-forge -c cadquery python=3.10 cadquery=master vtk=9.2.2`

so I had to relax the dependencies: 
```
mamba create -n jcq3 -c conda-forge -c cadquery python cadquery=master vtk
mamba activate jcq3
```

then when installing the pip dependencies: 
`pip install jupyter-cadquery==3.5.2 cadquery-massembly==1.0.0 matplotlib`

it failed again:
[image]

cheers

bernhard-42, 2024-04-26:
I currently don't have time to update jupyter_cadquery. My focus is on https://github.com/bernhard-42/vscode-ocp-cad-viewer. So if you are willing to work with VS Code, this works great with build123d and cadquery. I plan to bring all the features of OCP CAD VIewer to Jupyter in the future, but I don't have an ETA.

bernhard-42, 2025-03-20:
fyi, I am now working on the integration of the latest viewer capabilities from ocp_vscode and the latest cadquery/build123d.
cad-viewer-widget is ported, now I'll integrate it into jupyter-cadquery
Hope this issue will be addressed by the new version

bernhard-42, 2025-04-18:
Please try the latest Jupyter Cadquery 4.0.2. Installation according to README

Note: When you want to install both CadQuery and build123d, I think you need to use cadquery 2.6-dev to avaoid conflicts with build123d