bernhard-42/jupyter-cadquery #110: Please tag releases on git/GitHub

KilowattSynthesis, 2024-11-11:
Please tag releases that get published to the PyPi package manager on GitHub so that comparing between versions, and matching release versions to source code, is much easier. It only takes a second! 

It looks like the last tag is v1, which was a while ago.

bernhard-42, 2025-03-20:
will do with the next release

KilowattSynthesis, 2025-04-18:
It looks to me like you released the 1.0.0 version in 2023, after all the 3.x.x versions in 2022, and that you had "Latest release" checked on the 1.0.0 version, thus making it look like you hadn't tagged a version since 1.0.0. 

All seems fine to me! Feel free to close. Awesome to see new releases of this coming out.

closed 2025-04-18 by bernhard-42