bernhard-42/jupyter-cadquery #113: Incompatible numpy dependency vs. build123d 0.8.0

pmuller, 2024-12-02:
build123d 0.8.0 now requires numpy `>=2,<3`, but some of jupyter-cadquery's dependencies seem to require numpy `<2`:

```
$ jcv
[JCV] Creating a Jupyter kernel specification called 'jcv' for this conda environment
Installed kernelspec jcv in /home/foo/.local/share/jupyter/kernels/jcv
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File ".venv/lib/python3.12/site-packages/jupyter_cadquery/__init__.py", line 34, in <module>
    from .ocp_utils import occt_version
  File ".venv/lib/python3.12/site-packages/jupyter_cadquery/ocp_utils.py", line 14, in <module>
    from cadquery import Compound, Location, Color
  File ".venv/lib/python3.12/site-packages/cadquery/__init__.py", line 22, in <module>
    from .occ_impl import exporters
  File ".venv/lib/python3.12/site-packages/cadquery/occ_impl/exporters/__init__.py", line 10, in <module>
    from ...cq import Workplane
  File ".venv/lib/python3.12/site-packages/cadquery/cq.py", line 62, in <module>
    from .sketch import Sketch
  File ".venv/lib/python3.12/site-packages/cadquery/sketch.py", line 28, in <module>
    from .occ_impl.sketch_solver import (
  File ".venv/lib/python3.12/site-packages/cadquery/occ_impl/sketch_solver.py", line 3, in <module>
    from nptyping import NDArray as Array
  File ".venv/lib/python3.12/site-packages/nptyping/__init__.py", line 30, in <module>
    from nptyping.ndarray import NDArray
  File ".venv/lib/python3.12/site-packages/nptyping/ndarray.py", line 32, in <module>
    from nptyping.shape_expression import (
  File ".venv/lib/python3.12/site-packages/nptyping/shape_expression.py", line 35, in <module>
    from nptyping.typing_ import Literal
  File ".venv/lib/python3.12/site-packages/nptyping/typing_.py", line 66, in <module>
    Bool8: TypeAlias = np.bool8
                       ^^^^^^^^
  File ".venv/lib/python3.12/site-packages/numpy/__init__.py", line 410, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'bool8'. Did you mean: 'bool'?
```

For reference:
```
$ uv tree
cad v0.1.0
├── build123d v0.8.0
│   ├── anytree v2.12.1
│   │   └── six v1.16.0
│   ├── cadquery-ocp v7.7.2
│   ├── ezdxf v1.3.4
│   │   ├── fonttools v4.55.0
│   │   ├── numpy v2.0.2
│   │   ├── pyparsing v3.2.0
│   │   └── typing-extensions v4.12.2
│   ├── ipython v8.30.0
│   │   ├── decorator v5.1.1
│   │   ├── jedi v0.19.2
│   │   │   └── parso v0.8.4
│   │   ├── matplotlib-inline v0.1.7
│   │   │   └── traitlets v5.14.3
│   │   ├── pexpect v4.9.0
│   │   │   └── ptyprocess v0.7.0
│   │   ├── prompt-toolkit v3.0.48
│   │   │   └── wcwidth v0.2.13
│   │   ├── pygments v2.18.0
│   │   ├── stack-data v0.6.3
│   │   │   ├── asttokens v3.0.0
│   │   │   ├── executing v2.1.0
│   │   │   └── pure-eval v0.2.3
│   │   └── traitlets v5.14.3
│   ├── numpy v2.0.2
│   ├── ocpsvg v0.3.1
│   │   ├── cadquery-ocp v7.7.2
│   │   └── svgelements v1.9.6
│   ├── py-lib3mf v2.3.1
[... 156 lines cut ...]
│   │   ├── numpy v2.0.2
│   │   ├── pyparsing v3.2.0
│   │   └── traitlets v5.14.3
│   ├── numpy-quaternion v2024.0.3
│   │   ├── numba v0.60.0
│   │   │   ├── llvmlite v0.43.0
│   │   │   └── numpy v2.0.2
│   │   ├── numpy v2.0.2
│   │   └── scipy v1.14.1 (*)
│   ├── voila v0.3.8
│   │   ├── jupyter-client v7.4.9 (*)
│   │   ├── jupyter-core v5.7.2 (*)
│   │   ├── jupyter-server v1.24.0 (*)
│   │   ├── jupyterlab-server v2.27.3 (*)
│   │   ├── nbclient v0.5.13 (*)
│   │   ├── nbconvert v6.5.4 (*)
│   │   ├── traitlets v5.14.3
│   │   └── websockets v14.1
│   └── webcolors v1.13
├── lxml[html-clean] v5.3.0 (*)
├── basedpyright v1.22.0 (group: dev)
│   └── nodejs-wheel-binaries v22.11.0
└── ruff v0.8.1 (group: dev)
(*) Package tree already displayed
```

bernhard-42, 2025-03-20:
fyi, I am now working on the integration of the latest viewer capabilities from ocp_vscode and the latest cadquery/build123d.
cad-viewer-widget is ported, now I'll integrate it into jupyter-cadquery.
This will address this issue

bernhard-42, 2025-04-18:
Solved with 4.0.2 as long as you use cadquery 2.6.0dev (master)

closed 2025-04-18 by bernhard-42